HesaiTechnology/HesaiLidar_General_ROS #6: Problem with publishing topic /raw points

remitoudic, 2020-01-16:
Hey , 
I got the chance to test a Pandar 64 .

yangfanFred, 2020-01-17:
hihi ：）

First of all, try using our PandarView software to connect Lidar so to make sure there's no problem with the Lidar itself. Please see our user mannual about how to connect Lidar and use PandarView.

Also please provide the Lidar settings web page information and Sever ip setting you are using.

remitoudic, 2020-01-22:
Thank you  Yangfang . 
Unfortunately  I don't have the Lidar anymore but could test it with windows software and it was working     :+1: 
Just a remark for the people of Hesai , the windows software given on the usb stick with the lidar was outdated  and did not work for the Pandar 64 .  So it someone has the same issue download the new version  directly from the  Hesai website

closed 2020-01-22 by remitoudic

yangfanFred, 2020-02-10:
Thanks a lot remitoudic