HesaiTechnology/HesaiLidar_General_ROS #7: This repo is ridiculous as an official driver

cslxiao, 2020-04-15:
As an official driver, this repo is far from qualified.

1. So many parameters set in the launch file are not even used in the code, except for 
> serverIp, lidarRecvPort, gpsPort

2. weird naming in the main function: pandora. (I know you are copying and pasting)

3. Most importantly, this code doesn't work for pandar40p.

To conclude, as an official driver provided by an international lidar manufacturer, this repo is astonishingly ugly.

yangfanFred, 2020-04-15:
Hi xiao：

：）
First, to help you out, I would like to know more details about the issue that you are facing with pandar40P, is there any error or warning reported or you can't see the point cloud output?

Then, thanks for the feed back about the code quaility. Due to the tight timeline and large workload we were facing before, some code may doesn't look so gracefully. We've understood the situation and we will start our imporving plan soon.  

Thanks.

cslxiao, 2020-04-15:
First, I would like to apologize for my harsh words. I expected Hesai to do a better job and I believe you can do much better.
When I use this driver for pandar40p with parameter lidar_type set, point clouds can be received but the coordinates are just not right. I have switched back to the Pandar40p_ros.

yangfanFred, 2020-08-04:
Hi xiao,

Thanks for your understanding, we did some improments recently on our github projects including achitecture, descriptions and code changes. And there are more improve items on going. We highly apreciate your suggestion and will keep improve.

Thanks,
Fred

closed 2020-08-04 by yangfanFred