HesaiTechnology/HesaiLidar_General_ROS #8: Error build source code

duclinhfetel, 2020-06-19:
## Environment:
* OS: ubuntu 20.04
* ROS version: ROS Noetic Ninjemys

## Change
* PCL on ubuntu 20.04 needs c++14 --> I changed CMakeLists.txt from c++11 to c++14
When i run catkin_make, I notice error boost.

I want to fix this problem. Could everyone help me?
Thanks

beltransen, 2020-07-10:
Same case here. I had to change the compiler version, but still, a _boost_ error appears. Here it is the screenshot of the error:
[image]

yangfanFred, 2020-07-17:
Hi duclinhfetel and beltransen,

Our prjects were mostly built and tested on Ubuntu16.04 & ROS Kinetic since this evnironment are the most common used and stable combo. We didn't offically support Ubuntu 20.04 yet, we will look into this requirement and give feedbacks very soon.

Thank you

closed 2020-10-29 by duclinhfetel