HesaiTechnology/HesaiLidar_General_ROS #11: Impossible to compile after clone

beltransen, 2020-07-10:
## Tested environments:
- OS: Ubuntu 18.04 & 20.04
- ROS version: ROS Melodic & Noetic
## Error
Apart from the compiling errors reported in https://github.com/HesaiTechnology/HesaiLidar_PandarGeneral_ROS/issues/8, there are some errors introduced in the latest commits. As can be seen in the image below, the code is trying to used an undeclared variable. This error exists in many functions, where the parameter is called **pkt** instead of **packet**. 

[image]

Apparently, it is just a variable naming issue. However, after replacing **packet** by **pkt** in the affected methods, a new error is thrown:

[image]

It would be desirable to have a reliable driver with periodical updates that offer support to OS and dependencies version upgrades. However, at least it should be mandatory that known issues are solved and **commits are tested before being pushed to the repository**.

yangfanFred, 2020-07-17:
Hi beltransen,

Our prjects were mostly built and tested on Ubuntu16.04 & ROS Kinetic since this evnironment are the most common used and stable combo. We didn't offically support Ubuntu 20.04 yet, we will look into this requirement and give feedbacks very soon.

Thank you

beltransen, 2020-07-20:
It looks like your recent commits solved the problem. I'm closing the issue.

closed 2020-07-20 by beltransen

yangfanFred, 2020-07-21:
Thanks beltransen