HesaiTechnology/HesaiLidar_General_ROS #12: pcap to rosbag

eros-ck, 2020-07-16:
Hello,

Does this driver support reading the pcap files recorded by PandarView to publish in ROS msg (sensor_msg/Pointcloud2)?

Thanks.

yangfanFred, 2020-07-17:
Hi eros-ck,

Currently this driver dont't support the feature you described while we've already working on it and will be released very soon.

Thanks

yangfanFred, 2020-08-04:
We achieved this function and  it's under final test stage, will be released soon.

w634307272, 2020-10-05:
> We achieved this function and it's under final test stage, will be released soon.

Has the function already done?

yangfanFred, 2020-10-05:
Hi w63,

The function has been released several weeks ago.
You can check the instruction, now we accept Pcap file as the input to pare data and publish into ROS system.
You can use the rosbag record tool to get a rosbag.

Thanks,
Fred

EyedBread, 2025-02-03:
Hi @yangfanFred ,

When I try to rosbag record a PCAP file that was recorded a month ago, the new timestamps are set to current system time, overriding the previous timestamp. Is this intended?

Best regards,
Bread