HesaiTechnology/HesaiLidar_General_ROS #13: PTP Synctime Offset

philRudd69, 2020-07-21:
Hi Hesai-Crew,

first of all: Thank you very much for this ROS driver. It helps a lot, but unfortunately we encounter problems when we want to the time of our PTP master clock to be used in header->stamp and in the timestamps of each and every point in the custom PPointCloud message. Instead of the master clock time we experience an offset in the timestamp of about 20 years. Is there a bug in the ROS-driver? How do we fix this?
Thank you :)
Cheers

Markus

yangfanFred, 2020-07-22:
Hi Markus,

Which product are you using and what's the firmware version of the unit? We will look into this issue.

Thanks,
Fred

philRudd69, 2020-07-23:
Hi Fred,

sorry for the late reply.
I am using the 
- Pandar64
- a version of this GIT repo from roughly a year ago (sorry, I can't specify it any better right now)
- Ubuntu 18.04
- ROS Melodic

yangfanFred, 2020-07-24:
Hi Markus,

I did some test at my side. Things you described  doesn't look like a ROS bug, it seems more like a configuration issue, I'm not sure if the time from PTP master has successfully locked into your Lidar. 

Please check the PTP status and  clock source setting from Lidar's web page, make sure Lidar time is tracking or locked and clock source is correctly selected
[image]
[image]

Then please use wireshark to monitor the UDP packets, see the Date&Time fields to check if the time is correct. 

Thanks，
Fred

philRudd69, 2020-07-27:
Hi Fred,
many thanks for your support. We will check your proposal on Friday. 
Hopefully we can solve the issue then. Until then: Thank you!
Cheers
Markus

shahaizi0, 2022-07-01:
![2022-07-01_16-21](https://user-images.githubusercontent.com/63791997/176855392-4fb7f2c4-beca-43e8-a575-06dbf48cc662.png

shahaizi0, 2022-07-01:
我想问一下，网络主页PTP显示free run怎么解决