HesaiTechnology/HesaiLidar_General_ROS #16: Wrong Time Stamp!!!

jessychen1016, 2020-09-02:
Hi, I got one Pandar64 Lidar here and the timestamp shown in both PandarView and ROS topic is wrong and misordered which the date is shown as 2017.3.2 (It is supposed to be 2020.9.2).
Pleeeeazzzzz tell me what is wrong with it and how to fix it.
Thanks in advance!

yangfanFred, 2020-09-14:
Hi Jessy,

Pandar64 Lidar can't get the true time itself, It will need an external clock source if you want to get the correct timestamp.
Currently it support GPS and PTP, so you will need to connect a GPS module through the GPS port on connection box or you will need to have a PTP Master device on the same LAN.

Thanks,
Fred

nonlinear1, 2022-03-08:
I think this kind of driver is not good as ouster and velodyne! We cann't use your lidar for slam.