HesaiTechnology/HesaiLidar_General_ROS #17: Don't see pointcloud /pandat

cedricfrance76, 2020-09-22:
Hello, 

I am a studient and I try, for a autonomous-car project, to communicate between a pandar40M and ROS melodic (ubuntu  18.04) (Virtual machine). 
I followed "https://github.com/HesaiTechnology/HesaiLidar_General_ROS" and when i launch "roslaunch hesai_lidar hesai_lidar.launch lidar_type:="Pandar40M"",  the programme seems to be stop after to have written " Parse Lidar Correction success !!!".  (see the attached picture)

When i ask "rostopic list", i see "/pandar" in the list. But when I ask "rostopic echo /pandar", nothing is displaying. 

[image]
When i use rviz, we can add display by topic '/pandar but we cannot change fixed frame to lidar_type. Just "map" is written but any lidar_type is selectable.

I try to do the same thing on ROS kinetic on ubuntu 16.04 and it is exactly the same thing.

Could you help me please?

yangfanFred, 2020-09-25:
Hi cedricfrance76,

There must be some settings or operation not matching, we will soon publish a tutorial video to guide the using process, I will updated you once published.
Also, if you need our remote help, please contact your sales man to get in touch with us.

Thanks,
Fred

cedricfrance76, 2020-10-05:
Hi yangfan Fred,

Thank you to answer me.

I think that i use a pandar40 in fact. I try to work with  https://github.com/HesaiTechnology/Pandar40_ROS<%20https:/github.com/HesaiTechnology/Pandar40_ROS> but in this case i have  a other issue.
When the node start, it stop immediatly. We see this message :

[hesai_lidar-2] process has died [pid 30125, exit code -11, cmd /home/cedric/rosworkspace/devel/lib/hesai_lidar/hesai_lidar_node __name:=hesai_lidar __log:=/home/cedric/.ros/log/66c8f036-070b-11eb-8f7d-080027f6e083/hesai_lidar-2.log].
log file: /home/cedric/.ros/log/66c8f036-070b-11eb-8f7d-080027f6e083/hesai_lidar-2*.log

And i didn’t find this file hesai_lidar-2*.log
With a «  cat  /home/cedric/.ros/log/66c8f036-070b-11eb-8f7d-080027f6e083/*.log » i see some details of the other file .log but nothing on what is the problem.

Have you a idea for this problem ?

Thank you.

Best regard

Provenance : Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows 10

Envoyé le :vendredi 25 septembre 2020 04:43
À : HesaiTechnology/HesaiLidar_General_ROS<mailto:HesaiLidar_General_ROS@noreply.github.com>
Cc : cedricfrance76<mailto:cedric.darre@hotmail.fr>; Author<mailto:author@noreply.github.com>

cedricfrance76, 2020-10-06:
Hi yangfan Fred,

I think I find a mistake in github program.
In the main.cc, there is not a « ! » in the condition to say that we communicate with Ethernet.
( line 50 in main.cc)

[cid:image002.png@01D69BC0.A3EC3240]

[cid:image006.png@01D69BC0.A3EC3240]

After the correction, there were no more bugs and I saw data in the topic and the imagecloud in rviz.

Thank you

De : Cedric DARRE<mailto:cedric.darre@hotmail.fr>
Envoyé le :lundi 5 octobre 2020 15:10
À : HesaiTechnology/HesaiLidar_General_ROS<mailto:reply@reply.github.com>
Objet :RE : [HesaiTechnology/HesaiLidar_General_ROS] Don't see pointcloud /pandar (#17)

Hi yangfan Fred,

Thank you to answer me.

I think that i use a pandar40 in fact. I try to work with  https://github.com/HesaiTechnology/Pandar40_ROS<%20https:/github.com/HesaiTechnology/Pandar40_ROS> but in this case i have  a other issue.
When the node start, it stop immediatly. We see this message :

[hesai_lidar-2] process has died [pid 30125, exit code -11, cmd /home/cedric/rosworkspace/devel/lib/hesai_lidar/hesai_lidar_node __name:=hesai_lidar __log:=/home/cedric/.ros/log/66c8f036-070b-11eb-8f7d-080027f6e083/hesai_lidar-2.log].
log file: /home/cedric/.ros/log/66c8f036-070b-11eb-8f7d-080027f6e083/hesai_lidar-2*.log

And i didn’t find this file hesai_lidar-2*.log
With a «  cat  /home/cedric/.ros/log/66c8f036-070b-11eb-8f7d-080027f6e083/*.log » i see some details of the other file .log but nothing on what is the problem.

Have you a idea for this problem ?

Thank you.

Best regard

Provenance : Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows 10

yangfanFred, 2020-10-09:
Hi cedricfrance76,

Thanks a lot, now we've fixed the bug.

Best Regards,
Fred

closed 2020-10-09 by yangfanFred

CharlesHuan, 2023-03-23:
> Hi cedricfrance76,
> 
> There must be some settings or operation not matching, we will soon publish a tutorial video to guide the using process, I will updated you once published. Also, if you need our remote help, please contact your sales man to get in touch with us.
> 
> Thanks, Fred

Hi Fred, I meet the same question like this,I think I need a video tutorial or an illustrated tutorial .Where can I see the video tutorial?T-T