HesaiTechnology/HesaiLidar_General_ROS #18: 不能正常解析pcap文件！

LRCJ, 2020-09-30:
Ubuntu18.4,ROS(Melodic),gcc 7.5.0,

工程目录结构
HesaiLidar_General_SDK/
buile/
main.cpp
CMakeLists.txt

main.cpp只有回调函数lidarCallback和主函数，
int main(int argc,char **argv)
{
	PandarGeneralSDK pandarGeneral(std::string("/home/user/Desktop/PCAP/PandarQT_Cars.pcap"), \
		lidarCallback, 0, 0, 1, std::string("PandarQT"));
	pandarGeneral.LoadLidarCorrectionFile(string("/home/user/Desktop/PCAP/PandarQTCorrectionFile.csv"));
	pandarGeneral.Start();
	while (true)
	{
		sleep(100);
	}

	return 0;
}

void lidarCallback(boost::shared_ptr<PPointCloud> cld, double timestamp)
{
	printf("Frame timestamp:%lf,Point Size:%ld\n",timestamp,cld->points.size());
}

输出结果只有三行
Parse Lidar Correction...
Frame timestamp:1575604314.808093,Point Size:102
Frame timestamp:1575604314.808315,Point Size:22337
Frame timestamp:1575604314.908207,Point Size:22408
^C

但我读取的这个pcap文件肯定不止3个PPointCloud，因此对pcap文件的读取存在问题，查不出原因？

yangfanFred, 2020-09-30:
Hi LRCJ，

建议：
1. 请用ROS Driver直接测试你的PCAP文件，确认读取pcap功能的好坏，然后对比ROS Driver，添加调试打印语句来调试你的代码
2. LoadLidarCorrectionFile不接受文件路径；只接受校准文件字节流
3. ROS Driver是对SDK接口用法的最好Demo，不清楚函数用法的请参照ROS Driver的调用方式

Thanks,
Fred

LRCJ, 2020-09-30:
> Hi LRCJ，
> 
> 建议：
> 
> 1. 请用ROS Driver直接测试你的PCAP文件，确认读取pcap功能的好坏，然后对比ROS Driver，添加调试打印语句来调试你的代码
> 2. LoadLidarCorrectionFile不接受文件路径；只接受校准文件字节流
> 3. ROS Driver是对SDK接口用法的最好Demo，不清楚函数用法的请参照ROS Driver的调用方式
> 
> Thanks,
> Fred

LRCJ, 2020-09-30:
用ROS倒是可以正常读取pcap文件，直接把HesaiLidar_General_ROS clone下来，用catkin_make编译，可以正常读取，但我就是不想通过ROS，lidarCallback回调函数里面直接获取了点云，我就可以直接处理了，不需要发布ROS话题，再写个ROS的程序订阅话题再处理。ROS这东西实在太臃肿了。

closed 2020-09-30 by LRCJ

LRCJ, 2020-09-30:
> Hi LRCJ，
> 
> 建议：
> 
> 1. 请用ROS Driver直接测试你的PCAP文件，确认读取pcap功能的好坏，然后对比ROS Driver，添加调试打印语句来调试你的代码
> 2. LoadLidarCorrectionFile不接受文件路径；只接受校准文件字节流
> 3. ROS Driver是对SDK接口用法的最好Demo，不清楚函数用法的请参照ROS Driver的调用方式
> 
> Thanks,
> Fred

用ROS倒是可以正常读取pcap文件，直接把HesaiLidar_General_ROS clone下来，用catkin_make编译，可以正常读取，但我就是不想通过ROS，lidarCallback回调函数里面直接获取了点云，我就可以直接处理了，不需要发布ROS话题，再写个ROS的程序订阅话题再处理。ROS这东西实在太臃肿了。

yangfanFred, 2020-09-30:
了解的，所以建议对比ROS的调用方法来debug你自己用SDK开发的代码，肯定是一些函数调用方式不对的，既然ROS读取没有问题，那么它的调用SDK的流程就可以作为一个很好的参照呢！

LRCJ, 2020-10-06:
> 了解的，所以建议对比ROS的调用方法来debug你自己用SDK开发的代码，肯定是一些函数调用方式不对的，既然ROS读取没有问题，那么它的调用SDK的流程就可以作为一个很好的参照呢！

找到原因了，在读取本地pcap文件时，实例化的PandarGeneralSDK的参数LidarType是不需要指定的，直接给空字符串""就行了。

LRCJ, 2020-10-06:
> 了解的，所以建议对比ROS的调用方法来debug你自己用SDK开发的代码，肯定是一些函数调用方式不对的，既然ROS读取没有问题，那么它的调用SDK的流程就可以作为一个很好的参照呢！

建议你们写好这些类、函数的使用说明，像这样要注意的地方应该特别说明。