HesaiTechnology/HesaiLidar_General_ROS #21: Multiple Instances of Driver

alperk-cvml, 2020-11-05:
We are planning to use multiple lidars in our project. However we cannot launch multiple instances of the driver because it transmits to fixed`/pandar` topic and to fixed tf frame (`Pandar40P` for example) cannot be changed.

It would really be appeciated if frame and topic could be parameterized in the launch file.

DK-sr, 2020-11-05:
You can use ros group tag with namespace attribute for using the same multiple lidars. Then each topic name will remap with defined namespace (ex) topic is /pandar and (namespace group1 ) => new topic name : /group1/pandar ).
Please check the below links for more detailed information.
http://wiki.ros.org/roslaunch/XML/group
https://answers.ros.org/question/229185/roslaunch-namespace-substitution-within-group-tags/

alperk-cvml, 2020-11-06:
Thank you. This solves the problem for topic names. However I could not solve the problem for frame_id. We have 3 PandarQT's, naturally in different locations, so I need them to publish to different `tf`s. 

I tried `tf_prefix` like in this example: https://answers.ros.org/question/223410/how-to-set-namespace-for-tf/  Unfortunately it does still publish to the same frame.

yangfanFred, 2020-11-06:
Hi alperk, 

currently the ROS project is only designed as a demo to driver single Lidar, you will need to look into the code and modify from code inside to support your request.
We got your request and will look into developing a easy way to cofig and drive multiple Lidars but this will take a while.

Thanks,
Fred