HesaiTechnology/HesaiLidar_General_ROS #23: CPU Usage

alperk-cvml, 2020-11-06:
Processes related to ROS node of the drivers consume more CPU time than expected. 

On our ARM64 based platform the nodes consume 50-60% CPU. Using 4 LIDARs will continuously block 2 CPU cores. In comparison another LIDAR driver uses 5-7%. 

Is this expected? Are there settings/other ways to reduce this usage?

Thanks in advance,

berkerlogoglu, 2020-11-18:
Any update on this? the other LIDAR referred in the previous message is Velodnye Puck 32MR. There is significant difference in the processing power required.

yangfanFred, 2020-11-18:
Which Lidar module are you using? Different Lidar module have different data rates, that might be they key factor. We can look into your isseu, you can contact your sales man to get in touch with us.

Thanks，
Fred

berkerlogoglu, 2020-11-25:
We are using Pandar 40p and Pandar QT.  Both are similar.