HesaiTechnology/HesaiLidar_General_ROS #24: PandarScan.h: No such file or directory

philRudd69, 2020-11-09:
Hi 

we experience the following error whilst compiling the newest version of this repo:

``` 
In file included from /catkin_ws/src/HesaiLidar_General_ROS/src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pandarGeneral_internal.cc:20:0:
/catkin_ws/src/HesaiLidar_General_ROS/src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pandarGeneral_internal.h:35:10: fatal error: hesai_lidar/PandarScan.h: No such file or directory
 #include "hesai_lidar/PandarScan.h"
```
We can only solve this by compiling the workspace a second time.
Any ideas?

Cheers

Markus

DK-sr, 2020-11-10:
I had the same issue and fixed it by adding one add dependency line in CMakeLists.txt.
Currently, I made pull requests about this. Please check [here](https://github.com/HesaiTechnology/HesaiLidar_General_ROS/pull/22/commits/7aae245c970af3fa294cee49221a39dfe2cca6fe).

yangfanFred, 2020-11-11:
Thanks guys,

We will review and test the code

Thanks，
Fred

Shelfcol, 2021-01-17:
this answer works for me.  https://github.com/HesaiTechnology/HesaiLidar_General_ROS/commit/7aae245c970af3fa294cee49221a39dfe2cca6fe

    src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pcap_reader.cpp
    src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pandarGeneral.cc
)

(add this!!!!)
add_dependencies(PandarGeneral hesai_lidar_generate_messages_cpp)

target_include_directories(PandarGeneral PRIVATE
    src/HesaiLidar_General_SDK/src/PandarGeneralRaw/include
    src/HesaiLidar_General_SDK/src/PandarGeneralRaw/