HesaiTechnology/HesaiLidar_General_ROS #26: No Data with Custom Azimuth Field of View

Benjamin-Tan, 2020-12-05:
When I modify the Azimuth Field of View settings from the Web Control of Ending to be less than 360, no data (packets/pointclouds) is being published.

It is working fine when:
Start: 0~359
End: 360

It is not working fine when:
Start: 0
End: 1~359

yangfanFred, 2020-12-06:
Hi Benjamin,

Have a look at the description from readme, seems like you didn't configure the start_angle in launch file. Default setting is 0.
 
start_angle: Driver will publish one frame point cloud data when azimuth angel step over start_angle, make sure set to within FOV

Thanks，
Fred

Benjamin-Tan, 2020-12-06:
Hi Fred,

If I set the Azimuth FOV of start: 0, end: 270. The launch file start angle should remain as 0 isnt it?

yangfanFred, 2020-12-07:
That's right, any value from 0 ~270 should be fine.
I just tested using a Pandar40P Lidar.
If you are still facing the issue, contact your sales man and you can reach our FAE team to get detailed support.

Thanks,
Fred

Benjamin-Tan, 2020-12-08:
Hi Fred,

The Pandar40P is working fine for both cases. I was referring to PandarXT-32. I need to adjust my start angle in the launch file to the end angle in order to receive the data for PandarXT-32. 

E.g.
Start: 0
End: 270
Launch file start_angle: 270

closed 2020-12-08 by Benjamin-Tan