HesaiTechnology/HesaiLidar_General_ROS #28: Can't launch the lidar!!!

nickle-fang, 2021-01-05:
I follow the tutorial to build and configurate the source. But when I run`roslaunch hesai_lidar hesai_lidar.launch lidar_type:="PandarXT-32"`, I got the following error:

 ... logging to /home/nickle/.ros/log/4b5f7df8-461b-11ea-a086-2cf05d3dad57/roslaunch-nickle-3806.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: Invalid <node> tag: <node> tag has an empty 'ns' attribute. 

Node xml is <node name="hesai_lidar" ns="$(arg namespace)" output="screen" pkg="hesai_lidar" type="hesai_lidar_node">
		<param name="pcap_file" type="string" value="$(arg pcap_file)"/>
		<param name="server_ip" type="string" value="$(arg server_ip)"/>
		<param name="lidar_recv_port" type="int" value="$(arg lidar_recv_port)"/>
		<param name="gps_port" type="int" value="$(arg gps_port)"/>
		<param name="start_angle" type="double" value="$(arg start_angle)"/>
		<param name="lidar_type" type="string" value="$(arg lidar_type)"/>
		<param name="frame_id" type="string" value="$(arg frame_id)"/>
		<param name="pcldata_type" type="int" value="$(arg pcldata_type)"/>
		<param name="publish_type" type="string" value="$(arg publish_type)"/>
		<param name="timestamp_type" type="string" value="$(arg timestamp_type)"/>
		<param name="data_type" type="string" value="$(arg data_type)"/>
		<param name="lidar_correction_file" type="string" value="$(arg lidar_correction_file)"/>
	</node>
The traceback for the exception was written to the log file

yangfanFred, 2021-01-06:
Hi nickle-fang，

Try          roslaunch hesai_lidar hesai_lidar.launch lidar_type:="PandarXT-32" namespace:=lidar1

Thanks，
Fred

nickle-fang, 2021-01-06:
> Hi nickle-fang，
> 
> Try roslaunch hesai_lidar hesai_lidar.launch lidar_type:="PandarXT-32" namespace:=lidar1
> 
> Thanks，
> Fred

Thanks! That works. But when I launch the lidar, I got the following rostopic, NONE of it has a valid masage published.
/lidar1/pandar
/lidar1/pandar_packets
/rosout
/rosout_agg

shashankyld, 2023-09-06:
were you able to solve this issue?