HesaiTechnology/HesaiLidar_General_ROS #32: Why use the timestamp from the LiDAR time not the timestamp from roscore time?

harderthan, 2021-03-06:
Why use the timestamp from the LiDAR time?

https://github.com/HesaiTechnology/HesaiLidar_General_ROS/blob/035e54d767cf58983d70616f270950b3fb773852/src/main.cc#L95

The messages published by the lidar ros driver are often used by other nodes like localization, recognition.
At that time, a header time of lidar point message is necessary info to calculate latency or data interfolation, etc..
If the driver use the time from LiDAR sensor not roscore time, user should write codes to get rostime in duplicate.

I think, below code looks more good.

```cpp
pcl_conversions::toPCL(ros::Time::now(), cld->header.stamp);
```

harderthan, 2021-03-13:
Can I contribute to develop this issue for others?
If okay, I will create a new PR with this feature.

yangfanFred, 2021-03-13:
Hi haderthan...please try set the "timestmap_tye" in launch file to "realtime"

harderthan, 2021-03-22:
Thank. I made a new PR #35

closed 2021-03-22 by harderthan