HesaiTechnology/HesaiLidar_General_ROS #40: Support ROS2

harderthan, 2021-08-05:
Hello. 

I would like to discuss the support ros2 for HesaiLidar_General_ROS. 
**Is there a plan to support ros2?** I didn't find ros2 driver for PandarQT lidar from yours.

yangfanFred, 2021-08-11:
Hi Harderthan,

We have the plan to support ros2 and is wokring on it, will release once finished and fully tested.
Thanks for your contribute.

Thanks

harderthan, 2021-08-17:
Hello, @yangfanFred .
Thank you for the sharing about your plan to support `ros2` . 

I developed the ROS2 driver based on [the ROS1 driver repository](https://github.com/HesaiTechnology/HesaiLidar_General_ROS) you provide, with @joonr7.
You can check the code from [here](https://github.com/joonr7/HesaiLidar_General_ROS2/tree/dashing-devel).

I hope this is useful to you. Also, let me know if this code have some problem.

zhangyu-hesaitech, 2021-08-20:
Hello, @harderthan 
Thanks for your contribute.
I didn't find a way to use Time in ROS2 msg before，your code give a good example,it helps me a lot

harderthan, 2021-12-01:
Thanks! @zhangyu-hesaitech , @dadaroce

closed 2021-12-01 by harderthan