HesaiTechnology/HesaiLidar_General_ROS #43: buffer don't have space! when running with pcap file input

tianshiz, 2021-11-18:
I wanted to test out the pcap input using the sample pcap file from:
https://www.hesaitech.com/en/download/1?product=Pandar64
Next I launched the pcap:
`roslaunch hesai_lidar hesai_lidar.launch  pcap_file:="/root/hesai.pcap"`

Here is what I get:
`started core service [/rosout]
process[hesai/hesai_lidar-2]: started with pid [9880]
       ///////////////////////////////////////////////////////////////
       //     PandarGeneralSDK version: PandarGeneralROS_1.1.12      //
       ///////////////////////////////////////////////////////////////
set environment TZ=UTC0
Open correction file /root/hesai_ws/src/HesaiLidar_General_ROS/config/Pandar64.csv succeed
Parse Lidar Correction...
Load correction file from /root/hesai_ws/src/HesaiLidar_General_ROS/config/Pandar64.csv succeed
buffer don't have space!,1795
buffer don't have space!,3001
buffer don't have space!,5401
buffer don't have space!,7801
`

what does this mean? Am I feeding the pcaps incorrectly?

tianshiz, 2021-11-30:
this is happening randomly on the same pcap. sometimes the pcap works sometimes it doesnt

yangfanFred, 2021-12-03:
Hi tianshiz,

Please go through our readme for launching steps.
It seems like you didn't give correct lidar type,  eg. "lidar_type:="Pandar64""

Thanks,

tianshiz, 2021-12-03:
Hi Fred,

 Sorry I didn't update the post. I did eventually add the lidar_type:="Pandar64". But, even running w/ the sample data provided by Hesai for Pandar64, I've found the application to be unreliable. Sometimes it will work, sometimes it will give me this error.

Are you not seeing that w/ this data?

yangfanFred, 2022-03-28:
Hi tianshiz,

I downloaded the pcap file you provided and run roslaunch on my PC, monitored the log ouputs and point clouds on RIVZ.
I repeated this test for 10 times, there are no error message and the whole pcap file can be prcessed well.
Here are possible reasons I can think of about the issue on your end:
1. The PC you are using might be not fast enough, is it possible to change a better PC to test?
2. Is ther any other tasks added to the ROS driver code?  The issue might be caused by additional tasks added

Thanks,
Fred

tianshiz, 2022-03-28:
Thank you for taking the time to test that out. 
I am actually using a amazon aws cloud desktop with the following specs. I run a essentially barebones vanilla ros melodic docker image on it to run this driver.

Instance Size | # vCPUs | Memory (GiB) | Instance Storage (HDD) | Network Performance | Disk Read (MBps with 2 MiB block size)
-- | -- | -- | -- | -- | -- 
d2.2xlarge | 8 | 61 | 6 x 2 TB | High | 875 

the desktop uses  Intel Xeon E5-2676 v3 (Haswell) processors that run at a base clock frequency of 2.4 GHz and can deliver clock speeds as high as 3.0 GHz with Intel Turbo Boost.

As for your #2 question, the only thing that I do ontop of running this process is that I also start a rosbag to bag these pointcloud topics. Perhaps that's causing the issue?

yangfanFred, 2023-11-01:
Hi Tianshi,

This is highly posssible caused by the running environment,  you can try use a local PC to test.

Thanks
Fred