HesaiTechnology/HesaiLidar_General_ROS #47: PandarQT128 Driver Stack Smash

castaway2000, 2022-01-22:
i have received a demo unit of the PAndarQT 128 and am working to set up the drivers so far i have it installed however it seems there is no support for te QT128 in the general ROS drivers. When running this with the 128 and QT drivers i get the following.

`started roslaunch server http://metrolla-K700:45385/

SUMMARY
========

PARAMETERS
 * /hesai/hesai_lidar/coordinate_correction_flag: False
 * /hesai/hesai_lidar/data_type: 
 * /hesai/hesai_lidar/fixed_frame: 
 * /hesai/hesai_lidar/frame_id: Pandar128
 * /hesai/hesai_lidar/gps_port: 10110
 * /hesai/hesai_lidar/lidar_correction_file: /home/foobar/he...
 * /hesai/hesai_lidar/lidar_recv_port: 2368
 * /hesai/hesai_lidar/lidar_type: Pandar128
 * /hesai/hesai_lidar/multicast_ip: 
 * /hesai/hesai_lidar/pcap_file: 
 * /hesai/hesai_lidar/pcldata_type: 0
 * /hesai/hesai_lidar/publish_type: points
 * /hesai/hesai_lidar/server_ip: 192.168.1.201
 * /hesai/hesai_lidar/start_angle: 0.0
 * /hesai/hesai_lidar/target_frame: 
 * /hesai/hesai_lidar/timestamp_type: 
 * /rosdistro: noetic
 * /rosversion: 1.15.13

NODES
  /hesai/
    hesai_lidar (hesai_lidar/hesai_lidar_node)

auto-starting new master
process[master]: started with pid [279273]
ROS_MASTER_URI=http://localhost:11311/

setting /run_id to 8fba1064-7bb9-11ec-9c5b-5f43f2563725
process[rosout-1]: started with pid [279294]
started core service [/rosout]
process[hesai/hesai_lidar-2]: started with pid [279301]
       ///////////////////////////////////////////////////////////////
       //     PandarGeneralSDK version: PandarGeneralROS_1.1.12      //
       ///////////////////////////////////////////////////////////////
set environment TZ=UTC0
TcpCommandClientNew
TCP Command Client Init Success!!!
Load correction file from lidar
publishRawDataThread:set result [1]
publishRawDataThread:get thead 139726375347968, policy 0 and priority 0
buffer is: ,len is: -2067997732
tcpCommandClient_SendCmd
TcpCommand_buildHeader
tcpCommandReadCommand
tcpCommandHeaderParser
Parse Lidar Correction...
*** stack smashing detected ***: terminated
`

There is no messages that output and no logs that get produced. 

The launch file is default with the exception of the correction file wich i had updated to point to the pandar128 (non QT) correction file as there is no correction file yet for the QT128. I tested both the QT and 64 correction files too. 

I also shut off all other applications to validate this is not simply interfering with other running apps using resources and it persists. 

please advise.

yangfanFred, 2022-01-24:
Hi castaway2000,

For QT128, please use Pandar Swift ROS project.

Thanks，
Fred

castaway2000, 2022-01-24:
@yangfanFred i have the same issue on the Swift ROS project. I ran the build on that repo after testing a few ways to configure this. I will try to rebuild and test again.

castaway2000, 2022-01-24:
@yangfanFred i rebuilt it again and it seems to work now. thank you.

closed 2022-01-24 by castaway2000