HesaiTechnology/HesaiLidar_General_ROS #53: ROS2 branch build fails

ajay1606, 2022-02-16:
Hello, 

Anyone successfully built the ROS2 branch in galactic distribution?

Running command` colcon build --symlink-install` returns the following error message.

```
Starting >>> hesai_lidar
--- stderr: hesai_lidar                         
** WARNING ** io features related to openni will be disabled
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
CMake Error at /opt/ros/galactic/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name 'pcl_conversions' was
  not found before
Call Stack (most recent call first):
  CMakeLists.txt:92 (ament_target_dependencies)

---
Failed   <<< hesai_lidar [2.01s, exited with code 1]

Summary: 0 packages finished [2.09s]
  1 package failed: hesai_lidar
  1 package had stderr output: hesai_lidar
```

Any help is much appreciated.

cyp4x141, 2022-02-18:
try add this 
`find_package(pcl_conversions REQUIRED)`

ajay1606, 2022-02-20:
Hello @cyp4x141 , Thanks for your response.
As your suggestions i have included `find_package(pcl_conversions REQUIRED)` in the CMakeList but again build got failed with following error message;
```
--- stderr: hesai_lidar                         
** WARNING ** io features related to openni will be disabled
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
CMake Error at /opt/ros/galactic/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name 'tf2_geometry_msgs' was
  not found before
Call Stack (most recent call first):
  CMakeLists.txt:94 (ament_target_dependencies)

---
Failed   <<< hesai_lidar [3.09s, exited with code 1]
```
Hence i included `find_package(tf2_geometry_msgs REQUIRED)` this also in the CMakeList, but build failed with few more errors;

[image]

Any possible fix ?

cyp4x141, 2022-02-22:
before use any kind of PCL lidar, you should probably install pcap support by
`
$ sudo apt install libpcap-dev
$ sudo apt install libprotobuf-dev protobuf-compiler
`
as for prinary-expression problem, try change **CXX STANDARD** can fix most of them
`set(CMAKE_CXX_STANDARD 14)
`
to 
`set(CMAKE_CXX_STANDARD 17)
`

ajay1606, 2022-02-22:
@cyp4x141 Thanks for suggestions;

I have followed your suggestions; And confirmed following packages already installed.
```
$ sudo apt install libpcap-dev
$ sudo apt install libprotobuf-dev protobuf-compiler
```

And **CXX STANDARD** mofifed to  `set(CMAKE_CXX_STANDARD 17)`

But again error appears with 
[image]

And after including `find_package(tf2_geometry_msgs REQUIRED)` As following in CMakeLists & package.xml
```

find_package( Boost REQUIRED  COMPONENTS thread)
find_package(Boost REQUIRED COMPONENTS system)
# find_package(PCL REQUIRED)
find_package(ament_cmake REQUIRED)
find_package(ament_index_cpp REQUIRED)
find_package(rclcpp REQUIRED)
find_package(rclcpp_action REQUIRED)
find_package(rosidl_typesupport_c REQUIRED) 
find_package(rosidl_default_generators REQUIRED)
find_package(builtin_interfaces REQUIRED)
find_package(std_msgs REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(pcl_conversions REQUIRED)
find_package(tf2_geometry_msgs REQUIRED)
```

**Error:**
[image]

This is the error appears.

By the way, are you able to reproduce this error at your station ?
Please note my working environment : **Ubuntu 20.4/ ROS2 (Galactic)** 

Regards,
AJay

montagne52, 2022-02-28:
Hi AJay,

I am also aiming to use the Hesai LiDAR with **ROS2** (**Galactic**) on **Ubuntu 20.04**. In order for building to succeed, I modified the following parts:
* In `CMakeLists.txt`: 
[image]
* In `src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pandarGeneral_internal.h:`
[image]
* In `src/main_ros2.cc`:
[image]

Hence, adding `#include <tf2_ros/buffer.h>` to `src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pandarGeneral_internal.h` should make the build succeed. **However** there are still various warnings and I have **not** yet tested if this works fine with the target hardware (Pandar40p in my case). 

Interested to know what your results are with the target HW. 

Kind regards,
Robert

cyp4x141, 2022-03-01:
> 
adding #include <tf2_ros/buffer.h> to src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pandarGeneral_internal.h can make the build succeed.
as [montagne52](https://github.com/montagne52) say
HeSai support team told us that there is  no ros2 galactic support for PandarXT16  for now

ajay1606, 2022-03-01:
Hello, @montagne52 and @cyp4x141 Thanks both for your kind responses, now able to build without issues. Anyhow i will test this with our device (Hesai Pandar 64) and let you know.

Also, i have tested Padar 64 with this following ROS2 package and there were no issues observed as of now. Everything working with this ROS2 workspace.
https://github.com/tier4/hesai_pandar

Regards,
Ajay

ajay1606, 2022-03-04:
@cyp4x141 @montagne52, with the above modification able to get data without any issues. Thanks, both for valuable feedback. 

Regards,
Ajay

closed 2022-03-04 by ajay1606