HesaiTechnology/HesaiLidar_General_ROS #54: General config params of PandarXT-16

narutojxl, 2022-02-22:
Hello, authors,
I have a PandarXT-16,  could you please give some advice of  how to set the following params when i run with a slam system(lio-sam etc.), although i have read [readme doc](https://github.com/HesaiTechnology/HesaiLidar_General_ROS#details-of-launch-file-parameters-and-utilities).
>
    "pcldata_type":  0:mixed point clouds data type 1:structured point clouds data type

What's meaning of `mixed` and `structured` ?
>  
    "lidar_correction_file": "hesai_lidar_config_csvs/PandarXT-16.csv"
I found lidar_correction_file [PandarXT-16.csv](https://github.com/HesaiTechnology/HesaiLidar_General_ROS/blob/master/config/PandarXT-16.csv) in the config directory,  param `coordinate_correction_flag` should be `true` or `false`?

BTW, the frame id in this driver code is this picture 
[image]

Thanks for your help a lot!

ParkerChan, 2022-12-21:
Hello, I also have some questions to ask. Can I run lio-sam on this PandarXT-16? If it works, can you tell me which parameters need to be modified? Because I know that lio-sam needs point cloud data of "XYZIRT" type, but I'm not sure whether this Lidar has point cloud data of this format and whether the problem you mentioned has been solved? Thank you very much.

yangfanFred, 2022-12-22:
> 
1. they mean the structure of the output point cloud, mixed means a 1xN point array contains all points in a frame, structured means a 16xN point array(for XT16), each row contains points from a same channel.
2.  they are different correction method and will not influence each other. "coordinate_correction_flag" when set to 1, the calculation will need to take more CPU resoureces, we suggest to only set it to 1 when you have very enough processor resources.

yangfanFred, 2022-12-22:
> Hello, I also have some questions to ask. Can I run lio-sam on this PandarXT-16? If it works, can you tell me which parameters need to be modified? Because I know that lio-sam needs point cloud data [...]

We are not so familiar with lio-sam, but the required X, Y, Z, I, R, T information are all supported by the lidar and ROS driver.

Asagi-Lily, 2023-09-19:
Thanks a lot. May I ask how to change to outputting point clouds in XYZIR format？