HesaiTechnology/HesaiLidar_General_ROS #55: The timestamp is not good for this driver

nonlinear1, 2022-03-08:
I cann't use this driver for any usual slam such as, cartographer、loam、lio-samT. This timestamp for this driver cann't used for point cloud deskew,if no GPS or ptp devices.

zhangyu-hesaitech, 2022-05-06:
you can change the pointcloud type to velodyne type in lidarCallback

void lidarCallback(boost::shared_ptr cld, double timestamp, hesai_lidar::PandarScanPtr scan) // the timestamp from first point cloud of cld
{
if(m_sPublishType == "both" || m_sPublishType == "points"){
pcl_conversions::toPCL(ros::Time(timestamp), cld->header.stamp);
sensor_msgs::PointCloud2 cloud;
cloud.fields.clear();
cloud.fields.reserve(6);
cloud.height = 1;
cloud.width = (uint32_t)cld->points.size();
int offset = 0;
std::string name("x");
int count(1);
int datatype(sensor_msgs::PointField::FLOAT32);
offset = addPointField(cloud, name, count, datatype, offset);
name = "y";
count = 1;
datatype = sensor_msgs::PointField::FLOAT32;
offset = addPointField(cloud, name, count, datatype, offset);
name = "z";
count = 1;
datatype = sensor_msgs::PointField::FLOAT32;
offset = addPointField(cloud, name, count, datatype, offset);
name = "intensity";
count = 1;
datatype = sensor_msgs::PointField::FLOAT32;
offset = addPointField(cloud, name, count, datatype, offset);
name = "ring";
count = 1;
datatype = sensor_msgs::PointField::UINT16;
offset = addPointField(cloud, name, count, datatype, offset);
name = "timestamp";
count = 1;
datatype = sensor_msgs::PointField::FLOAT32;
offset = addPointField(cloud, name, count, datatype, offset);
cloud.point_step = offset;
cloud.row_step = 1 * cloud.point_step;
cloud.header.frame_id = cld->header.frame_id;
cloud.header.stamp = ros::Time(timestamp);
cloud.data.resize(cld->points.size() * cloud.point_step);
sensor_msgs::PointCloud2Iterator iter_x = sensor_msgs::PointCloud2Iterator(cloud, "x");
sensor_msgs::PointCloud2Iterator iter_y = sensor_msgs::PointCloud2Iterator(cloud, "y");
sensor_msgs::PointCloud2Iterator iter_z = sensor_msgs::PointCloud2Iterator(cloud, "z");
sensor_msgs::PointCloud2Iterator iter_intensity = sensor_msgs::PointCloud2Iterator(cloud, "intensity");
sensor_msgs::PointCloud2Iterator<uint16_t> iter_ring = sensor_msgs::PointCloud2Iterator<uint16_t >(cloud, "ring");
sensor_msgs::PointCloud2Iterator iter_timestamp = sensor_msgs::PointCloud2Iterator(cloud, "timestamp");
for(int i = 0; i < cld->points.size(); i++){
*(iter_x+i) = cld->points[i].x;
*(iter_y+i) = cld->points[i].y;
*(iter_z+i) = cld->points[i].z;
*(iter_intensity+i) = cld->points[i].intensity;
*(iter_ring+i) = cld->points[i].ring;
*(iter_timestamp+i) = cld->points[i].timestamp;
}
lidarPublisher.publish(cloud);

#ifdef PRINT_FLAG
printf("timestamp: %f, point size: %ld.\n",timestamp, cld->points.size());
#endif
}
if(m_sPublishType == "both" || m_sPublishType == "raw"){
packetPublisher.publish(scan);
#ifdef PRINT_FLAG
printf("raw size: %d.\n", scan->packets.size());
#endif
}
}

nonlinear1, 2022-05-06:
the timestamp from first point cloud of cld??? velodyne type timestamp is the time that is relative to first point of incoming point cloud. And the timestamp is very precise!

shibowing, 2022-05-13:
Dear the staff from HesaiTechnology, 

I notice that the timestamp for each lidar point is same value. May I ask is this issue fixed? 
For this "cld->points[i].timestamp" , May I ask  how to calculate the timestamp for each lidar points? Thank you very much!  
  
`sensor_msgs::PointCloud2Iterator iter_timestamp = sensor_msgs::PointCloud2Iterator(cloud, "timestamp");
for(int i = 0; i < cld->points.size(); i++){
*(iter_x+i) = cld->points[i].x;
*(iter_y+i) = cld->points[i].y;
*(iter_z+i) = cld->points[i].z;
*(iter_intensity+i) = cld->points[i].intensity;
*(iter_ring+i) = cld->points[i].ring;
*(iter_timestamp+i) = cld->points[i].timestamp;
}
lidarPublisher.publish(cloud);`