HesaiTechnology/HesaiLidar_General_ROS #60: How to Collect XYZIRT Type Point Datasets Using PandarXT-16

ParkerChan, 2022-12-21:
We want to use PandarXT-16 to collect XYZIRT type point cloud data. We wonder if this radar can support this kind of point cloud data information?
Hope to get your reply! thank you

yangfanFred, 2022-12-22:
src/HesaiLidar_General_SDK/src/PandarGeneralRaw/include/pandarGeneral/point_types.h

> We want to use PandarXT-16 to collect XYZIRT type point cloud data. We wonder if this radar can support this kind of point cloud data information? Hope to get your reply! thank you

Yeah, it's supported, you can find data type defination here in source code: src/HesaiLidar_General_SDK/src/PandarGeneralRaw/include/pandarGeneral/point_types.h

ParkerChan, 2023-05-27:
> src/HesaiLidar_General_SDK/src/PandarGeneralRaw/include/pandarGeneral/point_types.h
>
>
> Yeah, it's supported, you can find data type defination here in source code: [...]

> src/HesaiLidar_General_SDK/src/PandarGeneralRaw/include/pandarGeneral/point_types.h
>
>
> Yeah, it's supported, you can find data type defination here in source code: [...]

嗯嗯，后续工作中我也发现了点云数据格式支持，但现在又发现了一个问题，由于我的算法需要将雷达与惯性传感单元做时钟同步，我在阅读Pandar-xt16的底层代码时，发现代码中对点云的timestamp使用的不是ros::Time::now，而是另外一种时间戳，我想问的是，如果我在这个地方手动改为ros::time::now的话，会对点云数据有影响吗？感谢

Asagi-Lily, 2023-09-19:
你好，我使用的是XT32雷达，在/HesaiLidar_General_SDK/src/PandarGeneralRaw/include/pandarGeneral/point_types.h中也找到了格式，但目前输出的点云是XYZI格式，请问该如何修改以获得输出XYZIR格式的点云呢？感谢

yangfanFred, 2023-11-01:
Hi Parker,

ROS driver默认使用的时间戳为从雷达获取的雷达内部时间，反应的是精确的雷达采集到点的时间；
如果使用[ros time](ros::Time::now)，反应的是系统收到网络包的时间，没有本身的雷达时间精确；其他没有影响

Thanks,
Fred

yangfanFred, 2023-11-01:
Hi Lily,

目前输出的点云格式实际上是XYZITR格式，请再确认下，要修改也可以在这个头文件中修改的

Thanks,
Fred

Asagi-Lily, 2023-11-03:
非常感谢，我已经查看并确认输出的点云格式是XYZITR格式了