HesaiTechnology/HesaiLidar_General_ROS #61: Timestamp realtime source??

BaichenBadrakh, 2022-12-23:
I have Pandar64, using it for SLAM with Imu and other sensors. I need time synchronazition, what is timestamp realtime time source? Can i change the time source for ros::TIme::Now(), because time synchornize still has big problem for me.

yangfanFred, 2022-12-26:
> I have Pandar64, using it for SLAM with Imu and other sensors. I need time synchronazition, what is timestamp realtime time source? Can i change the time source for ros::TIme::Now(), because time synchornize still has big problem for me.

real time means use the packate receive system time as the timestamp for points, which is not accurate.
PTP or GPS is recommaned for time synchronazition.

BaichenBadrakh, 2022-12-26:
Thanks for reply, but there is very few information about PTP or GPS time synchronazition, can u share more information about how to do PTP or GPS for time source.

yangfanFred, 2022-12-26:
> Thanks for reply, but there is very few information about PTP or GPS time synchronazition, can u share more information about how to do PTP or GPS for time source.

which lidar model are you using? do you have the user mannual of the lidar unit? you can contact your salesman to get a copy of the user mannual, introductions about gps and ptp were included in the nuser mannual. As for ptp master, you can search the "PTP4L" tool, which can make your PC act as a PTP master to sync time of lidar.

BaichenBadrakh, 2023-01-06:
thanks for reply, close the issue

closed 2023-01-06 by BaichenBadrakh