HesaiTechnology/HesaiLidar_General_ROS #67: No install folder after catkin_make -DCMAKE_BUILD_TYPE=Release

danijel-fmci, 2023-02-20:
Hi,

I followed the steps from README.md, but after the build with `catkin_make -DCMAKE_BUILD_TYPE=Release` there is no install folder. Please take a look at the attached command history.
[HesaiLidar_General_ROS_terminal_commands_history.txt](https://github.com/HesaiTechnology/HesaiLidar_General_ROS/files/10785857/HesaiLidar_General_ROS_terminal_commands_history.txt)

However, I found another tutorial on YT on [Hesai Lidar Tutorial #3: Run ROS Driver with Connected Lidar](https://www.youtube.com/watch?v=Yx7oZm6ya3U) and after executing `catkin build --force-cmake`  I got install folder.
But still do not get the timestamp and point size in terminal output as per YT tutorial. 
Please see attached command history:
[[Youtube ]- Hesai Lidar Tutorial #3 Run ROS Driver with Connected Lidar_terminal_commands_history.txt](https://github.com/HesaiTechnology/HesaiLidar_General_ROS/files/10786135/Youtube.-.Hesai.Lidar.Tutorial.3.Run.ROS.Driver.with.Connected.Lidar_terminal_commands_history.txt)
[image]

The lidar is working properly using Pandar.exe.
[image]

The host OS is Windows 10, installed WSL1 and Ubuntu 18.04 with ROS melodic.
[host_spec.txt](https://github.com/HesaiTechnology/HesaiLidar_General_ROS/files/10786194/host_spec.txt)

Thank you.

Best,
Danijel

yangfanFred, 2023-02-23:
Hi Danijel,

uncomment print flag in main.c to enable output point info
[image]

danijel-fmci, 2023-02-27:
Hi @yangfanFred 

Now it outputs the timestamp and corresponding point size in the terminal.
Thank you very much.

Best,
Danijel

[image]

closed 2023-02-27 by danijel-fmci