HesaiTechnology/HesaiLidar_General_ROS #73: No install folder after catkin_make

tlsaudrl6to5, 2023-05-25:
hi. 

I've followed the instruction on github to install Hesai Lidar driver on Ubuntu 18.04

No error found while installing and catkin make. 

but after catkin_make, there is no install folder so that I couldn't find any configuration file (hesai_lidar.launch) 

when i run the comman "roslaunch hesai_lidar hesai_lidar.launch lidar_type:="Pandar64" frame_id:="Pandar64", no further step after the line "Load correction file from lidar succeed"

how could i fix this issue?

tlsaudrl6to5, 2023-05-25:
erase all folders in rosworkspace, and then build again followed by youtube, which catkin config --install, and then catkin build --force-cmake

now i have install folder, but when i run the command "roslaunch hesai_lidar hesai_lidar.launch lidar_type:="Pandar64" frame_id:="Pandar64", i got this error

RLException: [hesai_lidar.launch] is neither a launch file in package [hesai_lidar] nor is [hesai_lidar] a launch file name
The traceback for the exception was written to the log file

now how could i fix this issue?

tlsaudrl6to5, 2023-05-25:
Finally I made it! 

I'm not sure whether this procedure is correct or not, but anyway i finally plot point cloud with rviz

#step 1
catkin config --install
catkin build --force-cmake 

#step2
erase build and devel folders. 

#step3
catkin_make -DCMAKE_BUILD_TYPE=Release

#step4 
roslaunch hesai_lidar hesai_lidar.launch lidar_type:="Pandar64" frame_id:="Pandar64

#step5
open another terminal and type "rostopic list" and check whether topic is issued. 

#step6
run rviz and follow the instruction on Hesai Youtube. then you will see the point cloud

closed 2023-05-25 by tlsaudrl6to5

Addog666, 2024-11-26:
so have you found the configuration file hesai_lidar.launch? i have the same problem with you.

Addog666, 2024-11-26:
i have the same problem with you, thank you very much !!!