HesaiTechnology/HesaiLidar_General_ROS #77: Do I need to build HesaiLidar_General_SDK before?

shashankyld, 2023-06-22:
Do I need to build HesaiLidar_General_SDK before?

When building SDK, I had this following issue,

`In file included from /catkin_ws/src/HesaiLidar_General_ROS/src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pandarGeneral_internal.cc:20:0:
/catkin_ws/src/HesaiLidar_General_ROS/src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pandarGeneral_internal.h:35:10: fatal error: hesai_lidar/PandarScan.h: No such file or directory
 #include "hesai_lidar/PandarScan.h"`

bharadwajakshay, 2023-06-29:
As far as I can see in the CMakeLists, the SDK gets build when building the ROS driver. I did try to compile the SDK before and add it as a sub-directory, but then there are others errors w.r.t to calling the SDK class in the ROS driver.

yangfanFred, 2023-11-01:
Hi Shashank and Shay,

If you are using the ROS driver, SDK will be automatically built by the commands in readme file, no need to build SDK seperately.

Thanks,
Fred

Addog666, 2024-11-26:
i have the same problem with you, have you solved it?