HesaiTechnology/HesaiLidar_General_ROS #78: Compiling ROS2 branch with ROS2 Humble fails

bharadwajakshay, 2023-06-26:
Hi All,
I am trying to get the ros2 branch of the HesaiLidar_General _ROS to compile with ros2 Humble. currently I am getting the following issue. 

`HesaiLidar_General_ROS/src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pandarGeneral_internal.h:37:10: fatal error: tf2_ros/transform_listener.h: No such file or directory
   37 | #include <tf2_ros/transform_listener.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~` 

Upon checking, this seems to happen in the Hesai general sdk code of the repo. Has anyone faced the same issue? Any help towards fixing the issues is greatly appreciated. 

Changes already made :

-  added tf2, tf2_ros and tf2_geometry_msgs as REQUIRED
`find_package(tf2 REQUIRED)`
`find_package(tf2_ros REQUIRED)`
`find_package(tf2_geometry_msgs REQUIRED)`

- Added tf2, tf2_ros and tf2_geometry_msgs into target_link_libraries for PandarGeneralSDK based on the tutorial [here](https://docs.ros.org/en/humble/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Listener-Cpp.html)
`target_link_libraries(PandarGeneralSDK`
`PandarGeneral`
`${Boost_LIBRARIES}`
`${PCL_IO_LIBRARIES}`
`Boost::thread`
`pcap`
`tf2`
`tf2_ros`
`tf2_geometry_msgs)`

- Added tf2 and tf2_ros as build and exec dependency in package.xml
`<exec_depend>tf2</exec_depend>`
` <build_export_depend>tf2</build_export_depend>`
`<exec_depend>tf2_ros</exec_depend>`
` <build_export_depend>tf2_ros</build_export_depend>`

Nothing seems to work. I'm not sure where I am going wrong, or what I am missing. Anything really helps.

shashankyld, 2023-07-06:
I am having the same issue, tried the above steps but i still have this error:

#0 6.878 In file included from /workspace/rosworkspace/src/HesaiLidar_General_ROS/src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pandarGeneral_internal.cc:20:
#0 6.878 /workspace/rosworkspace/src/HesaiLidar_General_ROS/src/HesaiLidar_General_SDK/src/PandarGeneralRaw/src/pandarGeneral_internal.h:37:10: fatal error: tf2_ros/transform_listener.h: No such file or directory
#0 6.878    37 | #include <tf2_ros/transform_listener.h>

yangfanFred, 2023-11-01:
Hi Shay and Shashank,

We have launched ROS2.0 project, ROS2.0 versions are fully supported. Please try use the new project.

Thanks,
Fred

theinge9, 2024-04-09:
Hello Fred,

Does the ROS2.0 project mean to this package with ROS2 branch or that one [https://github.com/HesaiTechnology/HesaiLidar_ROS_2.0](url)?
I am confused in ROS2 driver for XT32.
Really looking forward to your advice.