HesaiTechnology/HesaiLidar_General_ROS #79: Launch issue in ROS Humble

NilaySener, 2023-07-17:
Hi,
I have a Hesai PandarQT and I am having some problems with the driver.

> OS: Ubuntu 22.04
> ROS: Humble

First of all, I solved the compilation problems I encountered with the PR [here.](https://github.com/HesaiTechnology/HesaiLidar_General_ROS/pull/76) But when I run the driver with the command `ros2 launch hesai_lidar hesai_lidar_launch.py` I get the following error:
```
[INFO] [launch]: All log files can be found below /home/nilay/.ros/log/2023-07-17-19-05-44-917593-nilay-19704
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [hesai_lidar_node-1]: process started with pid [19709]
[hesai_lidar_node-1]        ///////////////////////////////////////////////////////////////
[hesai_lidar_node-1]        //     PandarGeneralSDK version: PandarGeneralROS_1.1.10      //
[hesai_lidar_node-1]        ///////////////////////////////////////////////////////////////
[hesai_lidar_node-1] set environment TZ=UTC0
[hesai_lidar_node-1] Open correction file /home/nilay/workspace/hesai_ws/src/HesaiLidar_General_ROS/config/PandarQT.csv succeed
[hesai_lidar_node-1] Parse Lidar Correction...
[hesai_lidar_node-1] Load correction file from /home/nilay/workspace/hesai_ws/src/HesaiLidar_General_ROS/config/PandarQT.csv succeed
[ERROR] [hesai_lidar_node-1]: process has died [pid 19709, exit code -11, cmd '/home/nilay/workspace/hesai_ws/install/hesai_lidar/lib/hesai_lidar/hesai_lidar_node --ros-args -r __node:=hesai_node -r __ns:=/hesai --params-file /tmp/launch_params_gfvrqw2o --params-file /tmp/launch_params_09_tq0mi --params-file /tmp/launch_params_35l34eko --params-file /tmp/launch_params_4x6gxirr --params-file /tmp/launch_params_htdtq6qb --params-file /tmp/launch_params_fam0vef4 --params-file /tmp/launch_params_t9m_myu3 --params-file /tmp/launch_params_p1wz2u_r --params-file /tmp/launch_params_j9l461f9 --params-file /tmp/launch_params_cyjcv1l9 --params-file /tmp/launch_params__jzgeurf --params-file /tmp/launch_params_c0vrn5hd --params-file /tmp/launch_params_91jy_iak --params-file /tmp/launch_params__73aj107 --params-file /tmp/launch_params_bj4idyer --params-file /tmp/launch_params_zumly4k3'].
```
I would be very happy if you could help me with this.

jemmmel, 2023-08-31:
Any update with this issue?

yangfanFred, 2023-11-01:
Hi Nilay and Jemmmel,

We have launched ROS2.0 project, ROS2.0 versions are fully supported. Please try use the new project.

Thanks,
Fred