HesaiTechnology/HesaiLidar_General_ROS #81: buffer dont have space! How can i fix this???

KevinReny, 2023-08-26:


yangfanFred, 2023-11-01:
Hi Kevin,

This issue usually caused by incorrect lidar type or configration in launch file or some time consuming process added to the lidar callback function, please double check the application codes.

Thanks,
Fred

KevinReny, 2023-11-01:
好的已收到

antenna-fast, 2024-03-06:
the same problem occurs to me when I run the sdk in docker. How to solve?
the gdb info shows it currupted at:   

https://github.com/HesaiTechnology/HesaiLidar_General_SDK/blob/d7f39ee81534f21b7f69d376f8b77a6cc73eb1da/src/PandarGeneralRaw/include/pandarGeneral/pandarGeneral_internal.h#L362

Load correction file from /home/app/hesailidar_general_sdk/config/PandarXT-32.csv succeed
[New Thread 0x7f7e6e14c700 (LWP 540)]
[New Thread 0x7f7e6d94b700 (LWP 541)]
buffer don't have space!,3170
buffer don't have space!,6086
buffer don't have space!,9691
buffer don't have space!,11336
buffer don't have space!,12958
buffer don't have space!,15208
buffer don't have space!,16083
buffer don't have space!,17458
[Thread 0x7f7e6d94b700 (LWP 541) exited]
[New Thread 0x7f7e6d94b700 (LWP 542)]
All   145 pointClouds were read.
save frame cloud pcl start
[Thread 0x7f7e6d94b700 (LWP 542) exited]
save frame cloud pcl end
[Thread 0x7f7e6e14c700 (LWP 540) exited]
58			delete read_pcap_;
(gdb)

KevinReny, 2024-03-06:
好的已收到