HesaiTechnology/HesaiLidar_General_ROS #83: Obtain Lidar Serial Number

ryank-cobot, 2023-10-03:
Hello,

We are looking for a way to obtain the Serial Number of the lidar programmatically through the ROS driver. Is this capability possible? Thanks so much for your help in advance.

Best,
Ryan

yangfanFred, 2023-11-01:
Hi Ryan,

Serial Number can be obtained by sending lidar units TCP commands. 
ROS driver doesn't provide a function to directly obtain the SN, but it provides functions to send and receive TCP commands.
Please refer codes in "tcp_command_client.c" for General_ROS project,  please refer the codes in "ptc_client.cc" for ROS2.0 project. 
For details of the available TCP commands, please refer the TCP API manual for the lidar model you are using.

Thanks,
Fred

TimSchimansky, 2025-02-05:
This is probably not relevant anymore,but i have reverse engineered large parts of the Pandar HTTP API, which is also used to power the Webinterface. Therefore i can tell you, that at least for the Pandar64, it works as following:

`http://<lidar-ip>/pandar.cgi?action=get&object=device_info`

This will give you a JSON response with the fields:
- SN (Serial number)
- Mac
- SW_ver
- HW_ver
- FW_ver
- Model
- AngleOffset
- ProdDate
- ProdName

I think its a weird sign from Hesai to not just publish the HTTP API...