HesaiTechnology/HesaiLidar_General_ROS #84: Error code -11

hazcoper, 2023-11-06:
Hello, I am using the hesai pandar40p with this driver for ros1. I am using ubuntu 20.04 with ros noetic

When I try and launch the node i get the following error:

process[hesai/hesai_lidar-1]: started with pid [20654]
       ///////////////////////////////////////////////////////////////
       //     PandarGeneralROS version: PandarGeneralROS_1.1.15      //
       ///////////////////////////////////////////////////////////////
set environment TZ=UTC0
TcpCommandClientNew
TCP Command Client Init Success!!!
publishRawDataThread:set result [1]
publishRawDataThread:get thead 140236552173312, policy 0 and priority 0
Load correction file from lidar
Getting calibration from the device
Get calib lidar sdk buffer is: ,len is: 1566595908
handle 161151472 buffer 1566595920 len 1566595908
tcpCommandClient_SendCmd
[hesai/hesai_lidar-1] process has died [pid 20654, exit code -11, cmd /home/user1/user1/ws-as/devel/lib/hesai_lidar/hesai_lidar_node __name:=hesai_lidar __log:=/home/user1/.ros/log/aaeb10a4-7cf2-11ee-b013-c1301ebd7425/hesai-hesai_lidar-1.log].
log file: /home/user1/.ros/log/aaeb10a4-7cf2-11ee-b013-c1301ebd7425/hesai-hesai_lidar-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

however once every 10 or 15 tries, it will launch okay

yangfanFred, 2023-11-07:
Hi hazcoper，

Could be system related issue, could you please try the HesaiLidar_ROS_2.0 project? it supports ubuntu20.04 better, it supports most ROS1 and ROS2 versions.

Thanks,
Fred

hazcoper, 2023-11-07:
Okay, I will look into that
Could you tell me what are the improvements in HesaiLidar_ROS_2.0 and the new sdk?